thomaspasser/gpio-buttons #7: Fatal error

volumio, 2016-08-21:
Hi, I installed the plugin and overnight Volumio crashed with this error:

 /data/plugins/miscellanea/gpio-buttons/node_modules/brace-expansion/index.js:1
Aug 21 10:30:37 volumio volumio[5217]: (function (exports, require, module, __filename, __dirname) { &Ar
Aug 21 10:30:37 volumio volumio[5217]: ^
Aug 21 10:30:37 volumio volumio[5217]: SyntaxError: Unexpected token &
Aug 21 10:30:37 volumio volumio[5217]: at Object.exports.runInThisContext (vm.js:76:16)
Aug 21 10:30:37 volumio volumio[5217]: at Module._compile (module.js:513:28)
Aug 21 10:30:37 volumio volumio[5217]: at Object.Module._extensions..js (module.js:550:10)
Aug 21 10:30:37 volumio volumio[5217]: at Module.load (module.js:458:32)
Aug 21 10:30:37 volumio volumio[5217]: at tryModuleLoad (module.js:417:12)
Aug 21 10:30:37 volumio volumio[5217]: at Function.Module._load (module.js:409:3)
Aug 21 10:30:37 volumio volumio[5217]: at Module.require (module.js:468:17)
Aug 21 10:30:37 volumio volumio[5217]: at require (internal/module.js:20:19)
Aug 21 10:30:37 volumio volumio[5217]: at Object.<anonymous> (/data/plugins/miscellanea/gpio-buttons/node_modules/minimatch/minimatch.js:10:14)
Aug 21 10:30:37 volumio volumio[5217]: at Module._compile (module.js:541:32)

I have absolutely no idea on thiis...

klokslagB, 2016-08-21:
confirmed to be not working under 0.979
After updating to Volumio 0.979 (using system UI) I couldn't get my DAC working. After disabling the GPIO plugin and rebooting it worked. Will have a look at this tomorrow to reproduce.

tomatpasser, 2016-08-21:
This is weird! Seems like a module problem, maybe one needs to be updated.

macmpi, 2016-10-03:
Hi,
Did recent Volumio 2 updates ([like .998](gpio-buttons
)) fixed the issue, or does gpio-buttons need some update?
Hoping project is still active.

Thanks

volumio, 2016-10-03:
Yep, project is still alive... 
Did not test it with latest versions... Anyone?

tomatpasser, 2016-10-03:
Haven't had the time yet unfortunately. I will look on it sometime next week.

tomatpasser, 2016-10-14:
Just installed on the newest volumio release (2.0), will keep it running to see if problems come up.

tomatpasser, 2016-10-16:
I have had this running for two days now with no crashes. Is anyone still having this issue?

tomatpasser, 2016-10-26:
Open new issue if you still have this problem

closed 2016-10-26 by tomatpasser