thomaspasser/gpio-buttons #8: Can't install

5AMsan, 2016-10-09:


tomatpasser, 2016-10-14:
Can you try on the latest volumio release? (2.0)
If you have issues again, try without the DAC. There might be compability issues which I will look into.
Thanks

5AMsan, 2016-10-14:
I removed the DAC on the two last tests (sorry I forgot to mention).
I won't be able to test before monday as this device is in my office. Can I upgrade from GIT to test (I have remote acces to this RPI) ?

tomatpasser, 2016-10-14:
Upgrading with git is not guaranteed to be enough. You won't get the
changes made to the file system as specified in the "Build" volumio
repository.

5AMsan, 2016-10-14:
OK, so I'll try on monday with a fresh image.

5AMsan, 2016-10-21:
Sorry for delay...
Last install on 2.001 works fine !
Thank you.

closed 2016-10-21 by 5AMsan