thomaspasser/gpio-buttons #11: Volumio restarts when editing button config

macmpi, 2016-11-15:
Hi,

Installed latest gpio-buttons on rpi0 running Volumio 2.016.
When editing settings, Volumio fails and restarts, as per below:
```
Nov 15 17:28:24 volumio volumio[1338]: info: [{"prettyName":"GPIO Buttons","name":"gpio-buttons","category":"miscellanea","version":"0.7.5","icon":"fa fa-cube","enabled":true,"active":true},{"prettyName":"Spotify","name":"spop","category":"music_service","version":"1.2.0","icon":"fa-spotify","enabled":false,"active":true},{"prettyName":"Volumio Spotify Connect","name":"volspotconnect","category":"music_service","version":"1.3.8","icon":"fa-spotify","enabled":true,"active":true}]
Nov 15 17:29:54 volumio volumio[1338]: info: CoreCommandRouter::getUIConfigOnPlugin
Nov 15 17:29:54 volumio volumio[1338]: info: GPIO-Buttons: Getting UI config
Nov 15 17:29:54 volumio volumio[1338]: info: GPIO-Buttons: Empty config, loading defaults...
Nov 15 17:29:59 volumio volumio[1338]: info: CALLMETHOD: miscellanea gpio-buttons saveTriggers [object Object]
Nov 15 17:29:59 volumio volumio[1338]: info: CoreCommandRouter::executeOnPlugin: gpio-buttons , saveTriggers
Nov 15 17:29:59 volumio volumio[1338]: info: GPIO-Buttons: Applying config file...
Nov 15 17:29:59 volumio volumio[1338]: info: GPIO-Buttons: Play/pause on pin 2
Nov 15 17:30:00 volumio volumio[1338]: fs.js:640
Nov 15 17:30:00 volumio volumio[1338]: return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
Nov 15 17:30:00 volumio volumio[1338]: ^
Nov 15 17:30:00 volumio volumio[1338]: Error: EACCES: permission denied, open '/data/configuration/miscellanea/gpio-buttons/config.json'
Nov 15 17:30:00 volumio volumio[1338]: at Error (native)
Nov 15 17:30:00 volumio volumio[1338]: at Object.fs.openSync (fs.js:640:18)
Nov 15 17:30:00 volumio volumio[1338]: at Object.fs.writeFileSync (fs.js:1333:33)
Nov 15 17:30:00 volumio volumio[1338]: at Object.writeFileSync (/data/plugins/miscellanea/gpio-buttons/node_modules/jsonfile/index.js:110:13)
Nov 15 17:30:00 volumio volumio[1338]: at GPIOButtons.setConf (/data/plugins/miscellanea/gpio-buttons/index.js:151:5)
Nov 15 17:30:00 volumio volumio[1338]: at GPIOButtons.saveTriggers (/data/plugins/miscellanea/gpio-buttons/index.js:208:7)
Nov 15 17:30:00 volumio volumio[1338]: at CoreCommandRouter.executeOnPlugin (/volumio/app/index.js:818:29)
Nov 15 17:30:00 volumio volumio[1338]: at Socket.<anonymous> (/volumio/app/plugins/user_interface/websocket/index.js:479:38)
Nov 15 17:30:00 volumio volumio[1338]: at emitTwo (events.js:106:13)
Nov 15 17:30:00 volumio volumio[1338]: at Socket.emit (events.js:191:7)
Nov 15 17:30:00 volumio mpd[609]: client: [2] closed
Nov 15 17:30:00 volumio systemd[1]: volumio.service: main process exited, code=exited, status=1/FAILURE
Nov 15 17:30:00 volumio systemd[1]: Unit volumio.service entered failed state.
Nov 15 17:30:00 volumio systemd[1]: Starting dynamicswap.service...
Nov 15 17:30:00 volumio systemd[1]: Started dynamicswap.service.
Nov 15 17:30:00 volumio volumio[1496]: 512 MB or less RAM Detected, need to enable swap
Nov 15 17:30:00 volumio volumio[1496]: Enabling Swap
Nov 15 17:30:00 volumio volumio[1496]: swapon: /data/swapfile: swapon failed: Device or resource busy
Nov 15 17:30:00 volumio systemd[1]: dynamicswap.service: main process exited, code=exited, status=255/n/a
Nov 15 17:30:00 volumio systemd[1]: Unit dynamicswap.service entered failed state.
Nov 15 17:30:00 volumio systemd[1]: volumio.service holdoff time over, scheduling restart.
Nov 15 17:30:00 volumio systemd[1]: Starting dynamicswap.service...
Nov 15 17:30:00 volumio systemd[1]: Started dynamicswap.service.
Nov 15 17:30:00 volumio systemd[1]: Stopping Volumio Backend Module...
Nov 15 17:30:00 volumio systemd[1]: Starting Volumio Backend Module...
Nov 15 17:30:00 volumio systemd[1]: Started Volumio Backend Module.
Nov 15 17:30:00 volumio volumio[1504]: 512 MB or less RAM Detected, need to enable swap
Nov 15 17:30:00 volumio volumio[1504]: Enabling Swap
Nov 15 17:30:00 volumio volumio[1504]: swapon: /data/swapfile: swapon failed: Device or resource busy
Nov 15 17:30:00 volumio systemd[1]: dynamicswap.service: main process exited, code=exited, status=255/n/a
Nov 15 17:30:00 volumio systemd[1]: Unit dynamicswap.service entered failed state.
Nov 15 17:30:09 volumio volumio[1505]: info: -------------------------------------------
Nov 15 17:30:09 volumio volumio[1505]: info: -----            Volumio2              ----
Nov 15 17:30:09 volumio volumio[1505]: info: -------------------------------------------
Nov 15 17:30:09 volumio volumio[1505]: info: -----          System startup          ----
Nov 15 17:30:09 volumio volumio[1505]: info: -------------------------------------------
```
Thought config.json may not have proper rights as error message suggests, indeed:
`-rwxr-xr-x  1 volumio volumio   796 Nov 15 17:11 config.json
`
but chmoding it 777 did not fix the issue.
Thanks for any tip.

macmpi, 2016-11-15:
Actually it seemed that `/data/configuration/miscellanea/gpio-buttons/config.json` was empty and enclosing folder was `root:root`
Something wrong probably happened in the install process.

macmpi, 2016-11-15:
install.sh is run as root now, so this creates issues as config can not be saved anymore.
You would need to `chown volumio:volumio` `/data/configuration/miscellanea/gpio-buttons` and `/data/configuration/miscellanea/gpio-buttons/config.json`

tomatpasser, 2016-11-15:
Thanks for the pointers, looking at it now

tomatpasser, 2016-11-16:
Should be fixed in c45ad83. Could not reproduce crash but did notice permission problems - added chown in install script. Can you confirm it no longer crashes?

macmpi, 2016-11-16:
Yes fixes it.
I think if you had pre-existing config files, you would not see the bug, as they are not cleared with uninstall.
Indeed, you may want to have an `uninstall.sh` that removes configuration folder on script uninstall for complete cleanup.

closed 2016-11-18 by tomatpasser