thomaspasser/gpio-buttons #17: Volumio restarts when saving configuration

babzek, 2017-01-18:
Hello, 

first of all thanks for this plugin, looks cool! Unfortunately I cannot save configuration using the volumio UI, it restarts upon hitting save.
Volumio version: 2.041
Logs reporting plugin version as 0.7.6

here is volumio's log file when I hit save:
```
2017-01-18T18:04:40.302Z - info: CoreCommandRouter::getUIConfigOnPlugin
2017-01-18T18:04:40.432Z - info: Received Get System Version
2017-01-18T18:04:40.435Z - info: CoreCommandRouter::executeOnPlugin: system , getSystemVersion
2017-01-18T18:04:58.881Z - info: [{"prettyName":"GPIO Buttons","name":"gpio-buttons","category":"miscellanea","version":"0.7.6","icon":"fa fa-cube","enabled":true,"active":true},{"prettyName":"Spotify","name":"spop","category":"music_service","version":"1.2.0","icon":"fa-spotify","enabled":true,"active":true}]
2017-01-18T18:05:00.920Z - info: CoreCommandRouter::getUIConfigOnPlugin
2017-01-18T18:05:00.924Z - info: GPIO-Buttons: Getting UI config
2017-01-18T18:05:00.949Z - info: GPIO-Buttons: Empty config, loading defaults...
2017-01-18T18:06:20.816Z - info: CALLMETHOD: miscellanea gpio-buttons saveTriggers [object Object]
2017-01-18T18:06:25.161Z - info: -------------------------------------------
2017-01-18T18:06:25.173Z - info: -----            Volumio2              ----
2017-01-18T18:06:25.175Z - info: -------------------------------------------
2017-01-18T18:06:25.175Z - info: -----          System startup          ----
2017-01-18T18:06:25.176Z - info: -------------------------------------------
```
[..]

Is there any way to edit a file for the configuration?
I have tried the config.json file but it does not seem to do anything, ie file is modified but nothing shows up on UI and button does not work.

Thanks for reading.

Vincent

tomatpasser, 2017-01-19:
Hello, thanks for opening this issue!
Which platform are you using?

babzek, 2017-01-20:
I am using the Rpi 2 B along with a HiFiBerry amp+

tomatpasser, 2017-01-31:
Sorry for the wait. I've been looking at this today. Do you still have this issue?
Does it happen every time? Or only with some pin configurations?

I have not been able to reproduce the bug yet.

babzek, 2017-02-01:
Hi, not to worry I also do not have a lot of time!

I will be able to look more into this over the week-end or the course of 
next week.

So far I have been able to connect a rotary encoder to GPIO 6, 5 and 13 
and that worked ok along with a python script.

This seems to happen even if I do not have anything connected to the 
pins if I just click around the ui, configure some functions and click 
on save and it restarts.

I only have the spotify plugin alongside.

I'll try to install this on another Pi with no Amp to see if I can 
reproduce.

babzek, 2017-02-17:
Hello,

i have (finally) managed to do a fresh install on another Pi (with no Amp+) and everything is working normally. I will try to do a fresh install of everything on my main build but it'll take me a while. So in the meantime you can close this issue. If I can reproduce I'll let you know.

Thanks again for sharing your work, I look forward to the support of rotary encoder!

closed 2017-02-17 by babzek