thomaspasser/gpio-buttons #18: Play/Pause does not work properly

slidor, 2017-03-08:
I have a button attached to GPIO PIN 25.
I assigned the "Play/Pause" option.

It works three times:
1 btn press: play
2 btn press: pause
3 btn press: play
4 btn press: play (at least that is what is shown in the ui.

After stopping playback in the UI, I can start the game again ;)

tomatpasser, 2017-03-08:
Thanks for opening the issue. To investigate, can you report back on these details:
- Volumio version
- Media type (radio/local tracks)
- What happens to the music when you press the 4th time? Do you still hear it?

Thanks

slidor, 2017-03-08:
Hi, here is the desired information:

- Volumio Version2.118
- Media: web stream
- it doesn't matter how often i press the button, it keeps playing

slidor, 2017-03-10:
Hi,

I think I identified the problem: In your "Play/Pause" function you are toggeling the the state based on the current state, which you request via the web socket. It seems, that the player replies with "pause" instead of play. Thats the reason, why the plugin tries to "play" after the third button press. Furthermore, I think, the player does not support pausing, when playing a web stream. I will try to toggle play/stop instead of play/pause when I am back at home.

tomatpasser, 2017-03-12:
Hi again. I apologize for my late answer. Actually I am aware of this issue in volumio, see [https://github.com/volumio/Volumio2/issues/558](url). I wasn't aware that is publishes the state 'stop' when a web stream is stopped/paused!

Thanks for your solution, I will add it when I can. Or you can submit a pull request if you want.

tomatpasser, 2017-03-28:
Thanks, I have merged your pull request.

closed 2017-03-28 by tomatpasser