thomaspasser/gpio-buttons #24: Inconsistentent config read & write

macmpi, 2017-04-10:
It seems template config.json file (first below) is read fine at first init, and properly populates UIconfig default values.
```
{
  "playpause":{
    "enabled": {"value": false, "type": "boolean"},
    "pin": {"value": 17,"type":"number"},
    "value": {"value": 0, "type": "number"}
  },
  "volup":{
    "enabled": {"value": false, "type": "boolean"},
    "pin": {"value": 18,"type":"number"},
    "value": {"value": 0, "type": "number"}
  },
  "voldown":{
    "enabled": {"value": false, "type": "boolean"},
    "pin": {"value": 22,"type":"number"},
    "value": {"value": 0, "type": "number"}
  },
  "previous":{
    "enabled": {"value": false, "type": "boolean"},
    "pin": {"value": 23,"type":"number"},
    "value": {"value": 0, "type": "number"}
  },
  "next":{
    "enabled": {"value": false, "type": "boolean"},
    "pin": {"value": 24,"type":"number"},
    "value": {"value": 0, "type": "number"}
  },
  "shutdown":{
    "enabled": {"value": false, "type": "boolean"},
    "pin": {"value": 3,"type":"number"},
    "value": {"value": 0, "type": "number"}
  }
}

```
However once user saves config, it is then written a different format as per the one below:
```
{
  "playpause": {
    "enabled": {
      "value": true,
      "type": "boolean"
    },
    "pin": {
      "value": 17,
      "type": "number"
    },
    "value": {
      "value": 0,
      "type": "number"
    }
  },
  "volup": {
    "enabled": {
      "value": true,
      "type": "boolean"
    },
    "pin": {
      "value": 18,
      "type": "number"
    },
    "value": {
      "value": 0,
      "type": "number"
    }
  },
  "voldown": {
    "enabled": {
      "value": true,
      "type": "boolean"
    },
    "pin": {
      "value": 22,
      "type": "number"
    },
    "value": {
      "value": 0,
      "type": "number"
    }
  },
  "previous": {
    "enabled": {
      "value": true,
      "type": "boolean"
    },
    "pin": {
      "value": 23,
      "type": "number"
    },
    "value": {
      "value": 0,
      "type": "number"
    }
  },
  "next": {
    "enabled": {
      "value": true,
      "type": "boolean"
    },
    "pin": {
      "value": 24,
      "type": "number"
    },
    "value": {
      "value": 0,
      "type": "number"
    }
  },
  "shutdown": {
    "enabled": {
      "value": true,
      "type": "boolean"
    },
    "pin": {
      "value": 3,
      "type": "number"
    },
    "value": {
      "value": 0,
      "type": "number"
    }
  }
}
```
So, subsequent reads may not interpret the file properly anymore, and generate issues.
Write routine should probably be revised so that it generates `config.json` in the same original format (or, on opposite, `config.json` format be changed and read routine adapted accordingly).

tomatpasser, 2017-04-10:
I'm not sure I get this - have you had any issues with this?
I'm having trouble seeing the difference between your two formats apart from newline characters - afaik these are ignored is json, no?

macmpi, 2017-04-10:
I'm having an issue currently, and I'm wondering if it comes from this, but it may not (I tend to agree with your view).
I had saved the original config with defaults and restarted. Then modified 2 actions (play/pause 5 and shutdown 6) and restarted:
```
Apr 10 21:44:09 volumio volumio[1134]: PLUGIN START: gpio-buttons
Apr 10 21:44:09 volumio volumio[1134]: info: GPIO-Buttons: Applying config file...
Apr 10 21:44:09 volumio volumio[1134]: info: GPIO-Buttons: Play/pause on pin 5
Apr 10 21:44:09 volumio volumio[1134]: info: GPIO-Buttons: Vol+ on pin 18
Apr 10 21:44:10 volumio volumio[1134]: info: GPIO-Buttons: Vol- on pin 22
Apr 10 21:44:10 volumio volumio[1134]: info: GPIO-Buttons: Previous on pin 23
Apr 10 21:44:10 volumio volumio[1134]: info: GPIO-Buttons: Next on pin 24
Apr 10 21:44:10 volumio volumio[1134]: info: GPIO-Buttons: Shutdown on pin 6
Apr 10 21:44:10 volumio volumio[1134]: info: GPIO-Buttons started
```
Then made a change to restore play/pause on 17 and shutdown 3, and this lead to a as per below:
```
Apr 10 21:44:58 volumio volumio[1134]: info: [{"prettyName":"GPIO Buttons","name":"gpio-buttons","category":"miscellanea","version":"0.7.9","icon":"fa fa-cube","enabled":true,"active":false}]
Apr 10 21:45:02 volumio volumio[1134]: info: CoreCommandRouter::getUIConfigOnPlugin
Apr 10 21:45:02 volumio volumio[1134]: info: GPIO-Buttons: Getting UI config
Apr 10 21:45:20 volumio volumio[1134]: info: CALLMETHOD: miscellanea gpio-buttons saveTriggers [object Object]
Apr 10 21:45:20 volumio volumio[1134]: info: CoreCommandRouter::executeOnPlugin: gpio-buttons , saveTriggers
Apr 10 21:45:20 volumio volumio[1134]: info: GPIO-Buttons: Destroying trigger 0
Apr 10 21:45:20 volumio volumio[1134]: info: GPIO-Buttons: Destroying trigger 1
Apr 10 21:45:20 volumio volumio[1134]: info: GPIO-Buttons: Destroying trigger 2
Apr 10 21:45:20 volumio volumio[1134]: info: GPIO-Buttons: Destroying trigger 3
Apr 10 21:45:20 volumio volumio[1134]: info: GPIO-Buttons: Destroying trigger 4
Apr 10 21:45:20 volumio volumio[1134]: info: GPIO-Buttons: Destroying trigger 5
Apr 10 21:45:20 volumio volumio[1134]: info: GPIO-Buttons: Applying config file...
Apr 10 21:45:20 volumio volumio[1134]: info: GPIO-Buttons: Play/pause on pin 17
Apr 10 21:45:21 volumio volumio[1134]: info: GPIO-Buttons: Vol+ on pin 18
Apr 10 21:45:22 volumio volumio[1134]: fs.js:640
Apr 10 21:45:22 volumio volumio[1134]: return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
Apr 10 21:45:22 volumio volumio[1134]: ^
Apr 10 21:45:22 volumio volumio[1134]: Error: EACCES: permission denied, open '/sys/class/gpio/gpio18/value'
Apr 10 21:45:22 volumio volumio[1134]: at Error (native)
Apr 10 21:45:22 volumio volumio[1134]: at Object.fs.openSync (fs.js:640:18)
Apr 10 21:45:22 volumio volumio[1134]: at new Gpio (/data/plugins/miscellanea/gpio-buttons/node_modules/onoff/onoff.js:146:21)
Apr 10 21:45:22 volumio volumio[1134]: at GPIOButtons.applyConf (/data/plugins/miscellanea/gpio-buttons/index.js:187:12)
Apr 10 21:45:22 volumio volumio[1134]: at GPIOButtons.setConf (/data/plugins/miscellanea/gpio-buttons/index.js:146:7)
Apr 10 21:45:22 volumio volumio[1134]: at GPIOButtons.saveTriggers (/data/plugins/miscellanea/gpio-buttons/index.js:219:7)
Apr 10 21:45:22 volumio volumio[1134]: at CoreCommandRouter.executeOnPlugin (/volumio/app/index.js:959:29)
Apr 10 21:45:22 volumio volumio[1134]: at Socket.<anonymous> (/volumio/app/plugins/user_interface/websocket/index.js:489:38)
Apr 10 21:45:22 volumio volumio[1134]: at emitTwo (events.js:106:13)
Apr 10 21:45:22 volumio volumio[1134]: at Socket.emit (events.js:191:7)
Apr 10 21:45:22 volumio systemd[1]: volumio.service: main process exited, code=exited, status=1/FAILURE
```
It looks like GPIO 18 was not properly released (`EACCES: permission denied`) before re-applying new setting for that action (same GPIO18 pin).
After volumio restarts same issue happened with the next pin (GPIO 22).
Wondering what is the cause.
(I still probably have promise/defer issues also: plugin is always shown as inactive, even when it works)

While at it I noticed plugin is based on `onoff 1.1.1` (which depends on `epoll 0.1.19`), while Volumio is now built with `onoff 1.1.2`(which depends on `epoll 0.1.21`)

macmpi, 2017-04-11:
ok, it does not seem to happen anymore: maybe I had a bad character in some cosmetic change to UIConfig.json or something.
I still have the 'inactive" flag, but that's probably something else.
Closing.

closed 2017-04-11 by macmpi