thomaspasser/gpio-buttons #26: up-to-date plugin info

macmpi, 2017-04-12:
Would be nice if gpio-plugin info is kept up-to-date in master Volumio plugin view (versioning, description, etc).
You may want to routinely push PRs [there](https://github.com/volumio/volumio-plugins/blob/gh-pages/plugins/volumio/armhf/plugins.json#L51-L61) to keep it current.

tomatpasser, 2017-04-13:
Didn't know where to push it, thanks! Maybe it can be done automatically somehow, will think about that.

macmpi, 2017-04-15:
GPIO Plugin is listed within `system_controller` category in that Repo, whereas `package.json` defines it as `miscelleanea`.
You may want to line-up the 2 definitions by either changing category in `package.json` or in the repository.

tomatpasser, 2017-05-25:
Changed the definition to `system_controller` (29c424f) and made a pull request: https://github.com/volumio/volumio-plugins/pull/75

closed 2017-05-25 by tomatpasser

macmpi, 2017-05-25:
With that category change we may need to update `install.sh` older `config.json` [compatibility tweak](https://github.com/tomatpasser/gpio-buttons/blob/master/app/plugins/system_controller/gpio-buttons/install.sh#L6-L9), as new file will be in new location: need to add `mv` to right location after `seds` probably.
Can't do it right now, as I'm not in front of my Pi for few days to test: can you do it?

tomatpasser, 2017-05-25:
You are absolutely right. I figure it is best if script copies the file over if it exists in the old directory but no in the new directory. Then edit it afterwards with the new naming convention. I'll get right to it.

tomatpasser, 2017-05-25:
Fixed: f1420d8