thomaspasser/gpio-buttons #28: Extending GPIO actions

macmpi, 2017-04-13:
Looking at interesting @tim-felix-lange work in [GPIO amp on/off plugin](https://github.com/volumio/volumio-plugins/pull/50), I was wondering how much it would make sense to add this as a new action within gpio-buttons.
Keeping most GPIO features together in one plugin may have some interest to avoid conflicts should user run multiple GPIO-related plugins in parallel...

Maybe this could offer an interesting case to see how gpio-button could evolve to give more flexibility for creating other actions (settings, actions definitions, etc).

Any interest from authors?

tomatpasser, 2017-04-13:
When I began my work on this plugin, I wanted it to do everything related to the GPIO pins. A large number of actions, triggers and whatnot should be selectable for each pin.

To configure this, I would like to have a configuration interface like a list, where you can add each connection between the pin and the function. But as it stands you cannot make a list with variable number of rows in the current volumio framework. I think the framework is defined [here](https://github.com/volumio/Volumio2-UI/blob/dfb1e6c80087220f8f736d2028cf820d3b4805f8/src/app/plugin/plugin.controller.js). 

So ideally, we would create this new kind of interface element in volumio to be able to make this list. Unfortunately I haven't had the time to acquaint myself with how the interface is coded to be able to do this. I asked @volumio about it, but I think they are already overburdened with work on other parts of volumio.

Sorry for the long ramble, the essence is that I would love to have all this stuff in the plugin but I can't figure out how. As you know the current interface has a field for each action. We could just keep adding fields but I find it silly. What are your thoughts? 

ps. Ideally this plugin should be called just GPIO/GPIO controller or something, and people like @tim-felix-lange should add their work to this plugin instead of creating a new plugin for each function.

volumio, 2017-04-13:
Coud you explain what kind of list would you need?

tomatpasser, 2017-04-13:
@volumio A list where I could define an object with a number of elements (in this case a pin select, in/out select and function select). Then in the configuration I would have the option to add a new row in this list, a row for each object instance.  Like the alarm configuration.

tim-felix-lange, 2017-04-16:
Interesting idea @macmpi, considering dependencies and the possibility of multiple plugins trying to access the same GPIO, it does make quite a lot of sense to combine everything into one plugin. On the other hand, the sheer amount of possibilities that could be handled by GPIO would make it a very mighty plugin, which somehow contradicts the idea of small purpose-built plugins. Nevertheless, a group of people would definitely increase stability, code quality and maintainability. There are definitely some pros and cons. Overall I myself would slightly tend towards a unified plugin.

However, to realize such a plugin, we would need some major rework to the @volumio plugin backend, as @tomatpasser suggested, to make the plugin and its configuration more dynamic than it currently is (or seems?). Ideally we should be able to create objects of an abstract parent, that contain attributes like a set of allocated pins. This abstract class would then be implemented by a concrete peripheral, like button, an amplifier, a led, a RGB led, etc. Each of those concrete classes than has some additional attributes and methods that define its behavior: A button defines an input, has an active high/low selection and an associated action. An amplifier defines an output and activates a status-listener, and so on and so on. 
The configuration must be able to store a collection of those concrete peripherals, preferably a list, due to its implicit order, so we can recreate the settings page in the same order, but list is not a must have. 
The user interface must be able to reflect this dynamic creation and destruction of objects. I guess this might already be possible, somehow, but I couldn't find any docs on that, so I'd have to peek into some of the source files. 

Sorry for this lengthy post, but I think there is a lot to resolve if we want to make this happen.