thomaspasser/gpio-buttons #29: Inactive plugin status and multiple button presses

JonnyGee, 2017-04-23:
Hi, I'm having a few issues with the plugin all of a sudden, I've tested out various different combinations of my hardware and software but nothing seems to rectify the problem 100%

I'm using a HifiBerry Amp+ connected to a Pi2 using the latest Volumio (2.141) and your plugin installed directly from inside Volumio.

My system had a 6 button remote wired into the gpio pins, (I can't remember exactly which gpio as I have now dismantled the original remote but I know to avoid 2,3,18,19,20,21.) I have a shared "ground" daisy chained around each of my 6 buttons which is connected to a HIGH gpio, with all the others having their own individual LOW gpio.

This config worked perfectly until about 3 days ago when it started suffering from multiple button clicks from each different function, the only way to stop it was to disable the plugin. I have been doing regular "volumio pull" via SSH so I wonder if there is now a conflict with the plugin. (I have also noted the plugin no longer states it is active) However I have managed to get stable responses from my remote after removing the HifiBerry. 

I have also tried an rpi-update which results in kernel panic. Any help you could possibly give would be hugely welcome.

JonnyGee, 2017-04-24:
I have also noted that the plugin setup page is sometimes blank after changing gpio numbers on several occasions, requiring a full re-flash of Volumio.

tomatpasser, 2017-04-27:
I'm not sure what could be causing your problems. What exactly do you mean by "volumio pull" ? Have you been downloading the newest volumio commits manually with git, instead of using the internal update feature in volumio? 

@macmpi have contributed some changes recently which could help with the inactive plugin status. These improvements have been accepted to this repository, but not updated in the .zip file yet. I am planning to have the zip file updated automatically with the repository in the future. 

I will try and update the zip file soon, then you should try that version instead.

macmpi, 2017-04-27:
@tomatpasser understood and thanks.
I paused my WIP PR because I incidentally found some potential issues in Volumio pluginmanager and I'm looking into it more. Once that confirmed and eventually fixed if necessary, I'll further check GPIO Buttons to ensure consistency, and submit changes if need be.
I'll keep you posted within the PR.

JonnyGee, 2017-04-28:
Hi. Thanks for getting in touch. By Volumio Pull I mean the inbuilt command
to update the program directly, so no, not a manual update.

From reading up on other forums I think I have found that my problem was
caused by the high frequency signals and the way I was just testing a
couple of buttons at a time. Leaving some wires disconnected seemed to
introduce interference which was causing the extra button presses.

I look forward to seeing the new version. Thanks again.

Jon.

tomatpasser, 2017-04-28:
Yes, those wires might have picked up noise, they were acting as antennas if you don't connect them to anything.

tomatpasser, 2017-05-25:
The plugin status should be fixed as of #25.

closed 2017-05-25 by tomatpasser