thomaspasser/gpio-buttons #30: Can't install plugin

meaz, 2017-04-30:
Hi, 

I have volumio 2 installed on a pi zero. Everything works great! 

When I try to install gpio-buttons plugins, I get an erro. Here is what I have:
`Downloading plugin at http://127.0.0.1:3000/plugin-serve/b36982c8-f77f-4b20-abc1-3a02fc578994.zip
END DOWNLOAD: http://127.0.0.1:3000/plugin-serve/b36982c8-f77f-4b20-abc1-3a02fc578994.zip
Creating folder on disk
Unpacking plugin
Checking for duplicate plugin
Copying Plugin into location
Installing dependencies
The following error occurred when installing the plugin: Error`

I succeeded installing another plugin without any error.

Thanks!

tomatpasser, 2017-05-25:
Hello, sorry for not responding. 

I don't have a clue what could be causing this. 
Are you still having this problem?

meaz, 2017-06-11:
No, I restarted from scratch with the last volumio version and got no problem installing GPIO-Buttons... Weird...

tomatpasser, 2017-06-11:
Glad to hear, I'm closing this then.

closed 2017-06-11 by tomatpasser